riscv/riscv-aia #16: Request: Extension name (Sm*?) for CSR names support on toolchains

a4lg, 2022-05-22:
c.f.
https://wiki.riscv.org/pages/viewpage.action?pageId=327814

Much like Core-Local Interrupt Controller (CLIC) on fast interrupts TG, AIA specification adds number of CSRs (44 CSRs on 0.2.1-draft) for efficient yet flexible interrupt handling.

As a toolchain developer, that would be nice to have an extension name (possibly starting with `Sm`) for AIA to enable CSR names for AIA ... like CLIC (`Smclic` is shown above as an example ― WARNING: this name is not ratified).
If we have that (e.g. `smexample`), we can add it like `gcc -march=rv64gc_smexample ...` to enable AIA CSR names.

Of course, we can also use integer literal to control AIA CSRs (by number) and that (not defining the names in the toolchain) may be an option.  On the other hand, I think enabling AIA CSR names by default is not reasonable.

jhauser-us, 2022-06-19:
The AIA document still needs to be updated, but I can go ahead and tell you, for a machine-level environment, the Architecture Review Committee is proposing that all the registers and modified interrupt handling of the AIA will together be known as extension Smaia.

Note that the exact set of CSRs that the AIA adds depends on a few other factors:  the base ISA's XLEN (RV32 or RV64), whether S mode and the hypervisor extension are implemented, and whether the hart has an IMSIC.  But because these other factors are already specified separately, the different combinations can be inferred from the intersection of features.  There's no need to have an individual AIA extension name for each subset.

The compiler/development tools need not be concerned about whether an IMSIC exists but should just allow attempts to use the IMSIC CSRs if Smaia is specified.  Without an actual IMSIC, such attempts may trap, but that's not a problem for the development tools.

For a supervisor-level environment, the AR Committee is proposing to use the corresponding extension name Ssaia.  This is essentially the same as Smaia, except that the machine-level CSRs aren't visible at S level.

jhauser-us, 2022-08-30:
Resolved by commit 7292ba0.

closed 2022-08-30 by jhauser-us