riscv/riscv-aia #30: WARL values for hvictl.IID

JamesKenneyImperas, 2022-10-26:
Section 7.3.2 states that `hvictl.IID` is WARL. Can you please clarify what determines the legal values for this field, specifically:

1. whether there are any values in the standard interrupt range 0-15 that are always illegal, always legal or implementation-defined; and
2. what constraints are likely for values 16 and above.

Thanks.

jhauser-us, 2022-10-31:
Section 7.3.2. says
> IID is a WARL unsigned integer field with at least 6 implemented bits

The document might be too terse, but what it means is that there are some number of implemented bits for IID, a minimum of 6 and a maximum of 12 (because IID is bits 27:16 of `hvictl`).  If the number of implemented bits for IID is *k*, the valid values are all those representable in *k* bits, i.e., those in the range 0 to 2^*k* - 1.  That's the only reason the field is tagged as WARL.

JamesKenneyImperas, 2022-10-31:
Hi John,

I'd like further to clarify what happens when `hvictl.IID` is set to a value that corresponds to an undefined or inappropriate standard interrupt. For example, what happens if it is set to value 4 (the old `UTIE` of the deprecated N extension) or 7 (M-mode `MTIE`, which would be inappropriate for VS-mode). When `hvictl.VTI=1` and nothing else is pending, does this undefined or inappropriate interrupt get reported by `vstopi`? If some other interrupt is also pending, what priority would be used for this undefined or inappropriate interrupt in the highest-priority arbitration algorithm?

I was thinking that the WARL behavior of the `IID` field would exclude these cases, but from what you've said I'm not sure now that this is the case.

Thanks.

jhauser-us, 2022-10-31:
> For example, what happens if it is set to value 4 (the old `UTIE` of the deprecated N extension) or 7 (M-mode `MTIE`, which would be inappropriate for VS-mode). When `hvictl.VTI=1` and nothing else is pending, does this undefined or inappropriate interrupt get reported by `vstopi`?

It does, yes.  Software has the responsibility not to do something inappropriate.

> If some other interrupt is also pending, what priority would be used for this undefined or inappropriate interrupt in the highest-priority arbitration algorithm?

This is fully handled by the algorithm of Section 7.3.3, "Virtual supervisor top interrupt CSR (`vstopi`)".  Assume for your example that `hvictl` has fields VTI = 1 and IID = 4 or 7, and check the complete list of bullet items at the top of that section.  The only other possible candidate for the top interrupt in `vstopi` is number 9, SEI, from the first two bullets.  The priority number for that candidate SEI is either taken from `vstopei` (first bullet) or is 256 (second bullet), so there's no uncertainty.

JamesKenneyImperas, 2022-10-31:
Ok, thanks.

JamesKenneyImperas, 2022-10-31:
I'm sorry, I'm still trying to understand the full details here, and I think I closed this too early.

Is it possible for there to be a _tie_ in the indicated priority for an "undefined" virtual interrupt (e.g. when `VTI=1`, `ID=4`) and an SEI from the first two bullets? If so, how is the tiebreak performed, since the "undefined" virtual interrupt has no default priority specified for it?

Apologies in advance for my lack of understanding.

jhauser-us, 2022-11-03:
The problem with a virtual interrupt having a tie in priority number should be addressed by PR #33.

closed 2022-11-04 by JamesKenneyImperas