riscv/riscv-aia #31: vstopi behavior refers to vstopei

JamesKenneyImperas, 2022-10-26:
The behavior of CSR `vstopi` refers to the value of CSR `vstopei` (section 7.3.3). But `vstopei` is present only when an IMSIC is implemented. What happens if an IMSIC is not implemented?

Is there an assumption that `hstatus.VGEIN` can only be non-zero if an IMSIC is present? Is this stated anywhere?

Thanks.

jhauser-us, 2022-10-30:
Section 7.3.3, "Virtual supervisor top interrupt CSR (`vstopi`)", refers to `vstopei` in two bullet points:
> - if `hstatus`.VGEIN is the valid number of a guest interrupt file, bit 9 is one in both `vsip` and `vsie`, and `vstopei` is not zero ...
> - if `hstatus`.VGEIN is the valid number of a guest interrupt file, bit 9 is one in both `vsip` and `vsie`, and `vstopei` is zero ...

If a hart has no IMSIC, then there are no guest interrupt files for the hart, so it's hard to see how `hstatus`.VGEIN could be the "valid number of a guest interrupt file".  That makes both conditions false before we even get to worrying about the value of `vstopei` (which you are correct to point out doesn't exist in this scenario).  I don't see any logical ambiguity.

> Is there an assumption that `hstatus.VGEIN` can only be non-zero if an IMSIC is present?

Maybe it can be nonzero, but it can't be the number of a guest interrupt file when there are no guest interrupt files.

JamesKenneyImperas, 2022-10-31:
Ok, thanks.

closed 2022-10-31 by JamesKenneyImperas