riscv/riscv-aia #34: Virtualizing the contents of the IMSIC supervisor interrupt file

sorear, 2022-11-03:
Consider the M-mode-assisted virtualization paradigm:

* The H extension is (typically) not implemented.
* M-mode software switches between one or more S-mode worlds.
* At most one S-mode world is "privileged", runs with mstatus.TVM=0 and  mideleg[SEIP]=1.  This supports type-2 hypervisors with only a minimal   code footprint in M-mode.
* Limited support exists for assigning hardware directly to VMs; interrupts  are delivered to VMs by software, not hardware.

Without IMSIC, delivering an interrupt to a VM involves setting mvip[SEIP] (or the hidden mip[SEIP] bit pre-AIA), entering the guest, then using memory-access virtualization to handle the guest's accesses to (A)PLIC claim and complete registers.

With an IMSIC in the host, if there is a privileged S-mode world, the S-mode IMSIC must be under the control of the privileged world at all times because it can receive interrupts at any time.  But this prevents exposing a virtual IMSIC to the guest, because the guest needs to have its own view of (at minimum) the supervisor eipNN registers, and if Smstateen is not implemented all S-mode
worlds can observe the contents of eipNN.

Proposed solution: Mandate the functionality of mstateen[58] to be implemented by all harts which provide M-mode-assisted virtualization (support for the TVM, TW, and TSR bits in mstatus) and an IMSIC.  This could be either a dependency on the Smstateen extension, an alias for the state bit in some M-mode register, or a new state bit which provides the same functionality (logical AND if both are implemented).

jhauser-us, 2023-01-22:
Thanks for finding and reporting this issue!  Intended to be addressed by PR #37.  Please review.

jhauser-us, 2023-01-22:
Note that in PR #37, setting bit 9 of `mvien` (for SEIs) also blocks S-mode access to an IMSIC's supervisor-level interrupt file.