riscv/riscv-aia #44: Illegal instruction exception or virtual instruction exception?

vernonpang, 2023-06-06:
AIA spec:
> **Following the default rules for the hypervisor extension**, attempts from VS-mode to directly access a hypervisor or VS CSR, or from VU-mode to access any supervisor-level CSR (including hypervisor and VS CSRs), usually raise not an illegal instruction exception but instead a virtual instruction exception. For details, see the RISC-V Privileged Architecture.

Privileged spec:
> * in VS-mode or VU-mode, attempts to access an implemented non-high-half hypervisor CSR or VS CSR when the same access (read/write) would be allowed in HS-mode, assuming mstatus.TVM=0;
 > * in VU-mode, attempts to access an implemented non-high-half supervisor CSR when the same access (read/write) would be allowed in HS-mode, assuming mstatus.TVM=0;

Consider the following scenario
* `vsiselect=0x0`
* `siselect=0x0`
* VU-mode accesses `sireg`

Since said access is not HS-qualified (since `siselect=0x0`), should an illegal instruction exception instead of a virtual instruction exception be raised?

jhauser-us, 2023-06-11:
This should now be addressed with commits 5247b40 and 050473c.  The full document has been updated to 1.0-RC5.

vernonpang, 2023-06-11:
Appreciate the update!

vernonpang, 2023-06-13:
Reopened issue to clarify a scenario involving `vstopei`.

* `VU`/`VS`-mode
* CSR instruction accesses `vstopei` CSR directly by its own separate CSR address
* `hstatus.vgein=0` 

Said access is not HS-qualified since the same access in `HS`-mode will result in an illegal instruction access being raised. Should an illegal instruction exception or a virtual instruction exception be raised?

jhauser-us, 2023-06-16:
Believed to be fixed by #48.

@vernonpang, if you're willing, I'd be grateful if you could review Chapter 2 once more.  The window for fixing problems before ratification is closing fast.

vernonpang, 2023-06-22:
Chapter 2 looks good to me!

vernonpang, 2023-06-28:
I might have spotted something, but I want to check in with you first

It was mentioned in Chapter 2.1:
> CSR `mtopei` also exists only when an IMSIC is implemented.

I don't see the same specified for `stopei` and `vstopei`, but I assume it is implied.
If so, in Chapter 2.5:
> when the hart does not have an IMSIC ... despite that bit 58 of `mstateen0` is zero ... attempts from VS-mode to access `stopei` raise a virtual instruction exception

Should an illegal instruction exception be raised instead if `stopei` doesn't exist?

jhauser-us, 2023-06-30:
@vernonpang, I do appreciate you giving this chapter so much careful review.  Thank you!

However, in this case, the behavior you describe:
> when the hart does not have an IMSIC ... despite that bit 58 of `mstateen0` is zero ... attempts from VS-mode to access `stopei` raise a virtual instruction exception

is very much intentional.

Although our collection of RISC-V standards admittedly don't do a good job of explaining the multiple factors that go into such choices, the latest and greatest thinking on this subject says that what matters most here is not whether `stopei` "exists" but rather that extension Ssaia is "implemented".  That last fact gives us justification for assuming a hypervisor may want to emulate an IMSIC for a virtual machine.

Hopefully, my attempt at a brief explanation has been more illuminating than befuddling.

vernonpang, 2023-06-30:
Yes, that does make sense! I don't reckon I have any further updates for this. Thank you so much for taking the time to attend to my questions.

closed 2023-06-30 by vernonpang