riscv/riscv-aia #54: APLIC Domain Privilege

jrahmeh, 2023-10-16:
How do we tell the privilege (Machine or Supervisor) level of a domain.  The spec says that the root domain must be at Machine level and that a parent domain must also be at machine level.  This implies that supervisor must be a leaf.  But Machine can also be a leaf. So how do we identify a domain with supervisor privilege?

jhauser-us, 2023-10-18:
> a parent domain must also be at machine level. This implies that supervisor must be a leaf.

Not quite.  The spec says that the parent of a *supervisor-level* domain must be a machine-level domain.  Surprisingly, it's actually possible for the parent of a machine-level domain to be a supervisor-level domain.  Presumably, one would have this arrangement only when the machine-level child domain delivers interrupts to a different hart, or different set of harts, than the supervisor-level parent domain does.

But if we exclude that strange case, then yes, a supervisor-level domain must be a leaf domain.

> So how do we identify a domain with supervisor privilege?

This information is to be provided by means outside the hardware, such as the still-to-be-standardized RISC-V discovery mechanism that has been in development for some time.  For Linux, the information is in the device tree.

In recent years, there has been much discussion about how to give software the information it needs about the implemented RISC-V hardware.  The conclusion was to assume this function is fulfilled primarily by software-implemented standard data structures, such as Linux's device tree.

jrahmeh, 2023-10-20:
Thank you.  Closing.

closed 2023-10-20 by jrahmeh