riscv/riscv-aia #55: Is APLIC Child Index Relative or Absolute?

jrahmeh, 2023-10-16:
In section 4.5.2, is the child index field of sourcecfg relative or absolute?  Relative would be relative to the domain (first child is always at index 0), absolute would assign a unique index to each domain the hierarchy.  Can an interrupt be delegated to any child in the hierarchy or just to the immediate child?

jhauser-us, 2023-10-18:
>  Can an interrupt be delegated to any child in the hierarchy or just to the immediate child?

Using standard terminology for tree structures, each child has exactly one parent.  Section 4.2, which introduces interrupt domains, says:
> Each domain may then selectively delegate all or a subset of interrupt sources to its child domains in the hierarchy.

Nowhere is it suggested that a domain may delegate interrupts to the children of other domains, beyond the set of its own child domains.  A domain cannot even directly delegate to the children of its children (grandchildren, or further), only to its own immediate children.  "Delegate ... to its child domains" means exactly that, and nothing more.

Given that observation, the answer to your first question is in Section 4.5.2 ("Source configurations"):
> Child Index is a WLRL field that specifies the interrupt domain to which this source is delegated.  For an interrupt domain with *C* child domains, this field must be able to hold integer values in the range 0 to *C* − 1. Each interrupt domain has a fixed mapping from these index numbers to child domains.

That rules out the possibility of absolute numbers for the child index.

jrahmeh, 2023-10-20:
Thank you.  Closing.

closed 2023-10-20 by jrahmeh