riscv/riscv-aia #58: APLIC Level sensitive interrupts in MSI mode

gzaitd, 2023-11-16:
I have following queries on level sensitive interrupts in MSI mode.

> The pending bit is cleared whenever the rectified input value is low, when the interrupt is forwarded by MSI, or by a relevant write to an in clrip register or to clripnum.

Once level sensitive interrupt related rectified input value goes from low to high, MSI has to be forwarded by APLIC. However there is no specification on how much time APLIC can take to forward MSI. Since pending bit is cleared as soon as rectified input becomes low, Is it possible that interrupt is lost because before MSI is forwarded, pending bit is cleared, and due to clearing of pending bit, MSI is not generated.

How much time level sensitive interrupt wire has to remain asserted to successfully generate MSI.

> A write to a sourcecfg register will not by itself cause a pending bit to be cleared except when the source is made inactive. (But see Section 4.7.)

If we change sourcecfg.SM from Level0 to Level1 or vice-versa, rectified input value becomes low subsequently clearing the interrupt. Is this correct functioning and legal way to clear level sensitive interrupts?

jhauser-us, 2023-11-19:
> Since pending bit is cleared as soon as rectified input becomes low, Is it possible that interrupt is lost because before MSI is forwarded, pending bit is cleared, and due to clearing of pending bit, MSI is not generated.

That's how level-sensitive interrupts work:  If the incoming interrupt signal drops back low, then there's no longer a reason for an interrupt.  If you instead want the incoming interrupt signal to be remembered by the APLIC until an MSI is definitely sent, that's an edge-sensitive interrupt.  Configure the source mode as Edge1 or Edge0 instead of Level1/Level0.

> If we change sourcecfg.SM from Level0 to Level1 or vice-versa, rectified input value becomes low subsequently clearing the interrupt. Is this correct functioning and legal way to clear level sensitive interrupts?

The normal way to clear a level-sensitive interrupt is to do so at the source device as part of servicing the interrupt, not at the APLIC.  On the other hand, if you configure an interrupt's source mode as edge-sensitive, usually no action is needed to clear the interrupt at the APLIC, as it all happens automatically.

gzaitd, 2023-11-26:
Thanks for the reply !

closed 2023-11-26 by gzaitd