riscv/riscv-aia #59: vstopi behavior

umcann123, 2023-11-22:
The behavior of CSR vstopi is described in section 6.3.3.

> vstopi returns information about the highest-priority interrupt for VS level, found from among
> these candidates (prefixed by + signs):
> • if bit 9 is one in both vsip and vsie, hstatus.VGEIN is the valid number of a guest interrupt
> file, and vstopei is not zero:
> + a supervisor external interrupt (code 9) with the priority number indicated by vstopei;
> • if bit 9 is one in both vsip and vsie, hstatus.VGEIN = 0, and hvictl fields IID = 9 and
> IPRIO ̸= 0:
> + a supervisor external interrupt (code 9) with priority number hvictl.IPRIO;
> • if bit 9 is one in both vsip and vsie, and neither of the first two cases applies:
> + a supervisor external interrupt (code 9) with priority number 256;
> • if hvictl.VTI = 0:
> + the highest-priority pending-and-enabled major interrupt indicated by vsip and vsie
> other than a supervisor external interrupt (code 9), using the priority numbers assigned
> by hviprio1 and hviprio2;
> • if hvictl fields VTI = 1 and IID ̸= 9:
> + the major interrupt specified by hvictl fields IID, DPR, and IPRIO.

Does this imply that when hvictl fields VTI = 1 and IID != 9, our focus is solely on SEI (code 9) and the hvictl interrupt, disregarding other major interrupts?
Consider the following case:
**Only two pending and enabled interrupts exist: 44 and 43.
hvictl.VTI = 1, hvictl.IID = 44, and hvictl.IPRIO = 1. Additionally, all hviprio array values are zero.**
What is the expected/suggested behavior?
1. The return ID should be 44 because we need to consider other interrupts. (Default priority 44 > 43)  
2. The expected return ID should be 43. This is because, under the condition hvictl.VTI = 1 and hvictl.IID != 9, we solely prioritize SEI and the hvictl interrupt.

jhauser-us, 2023-11-25:
@umcann123, the list you quoted has five different possible candidates, each preceded by a necessary precondition.  For your example case, I believe the status of each of the five preconditions is:  1: false; 2: false; 3: false; 4: false; 5: true.  Hence, the only possible interrupt in this situation is the fifth candidate:
> the major interrupt specified by `hvictl` fields IID, DPR, and IPRIO.

Also note:  This fifth candidate in the list does not depend on the state of the corresponding bits in `vsip` and `vsie`.  For this candidate, either or both of those bits may be zeros; we don't care.

umcann123, 2023-11-25:
Hi @jhauser-us,

Thank you for your response! Just to clarify, in the case of the fifth candidate, are you suggesting that the only potential major interrupts considered are the SEI and hvictl.IID? Would this mean that other major interrupts are disregarded, even when both vsip and vsie are 1?

Thanks!

jhauser-us, 2023-11-25:
All those other non-SEI major interrupts from `vsip` and `vsie` are covered by the fourth candidate in the list.  However, the precondition for the fourth candidate is "`hvictl`.VTI = 0".  In your example, this is false, because you said VTI = 1.

Related to this same subject, the AIA spec has a comment that you may want to review:
> The list of candidate interrupts can be reduced to two finalists relatively easily by observing that the first three list items are mutually exclusive of one another, and the remaining two items are also mutually exclusive of one another.

umcann123, 2023-11-27:
@jhauser-us

Thanks for your reply!

closed 2023-11-27 by umcann123