riscv/riscv-aia #62: Question regarding mvip[9]

gzaitd, 2024-01-31:
1. Start with deleg[9]=0 mvien[9]=1 
2. An interrupt is delivered to S mode through setting mvip[9]=1
3. Spec says mvip is independent writeable bit of mip[9] and mip[9] is readonly and comes purely from IMSIC
4. We move to deleg[9]=1, mvien[9]=1
5. Now, should mip[9] see the '1' delivered in step 2 ?

jhauser-us, 2024-02-02:
There is no register named `deleg`.  Did you mean `mideleg`?

vithursons, 2024-03-24:
assuming `deleg `is `mideleg` , 5 is wrong.
assume **reg_meip** is software writable bit of **mip[9]** and **sig_seip** as **plic/aplic/imsic supervisor interrupt signal** 

mvien[9] | mideleg[9] | mip[9] | sip[9] | mvip[9]
-- | -- | -- | -- | -- 
0 | 0 | sig_meip \| reg_seip | read only 0 | reg_seip
0 | 1 | sig_meip \| reg_seip | sig_meip \| reg _seip  | reg_seip
1 | 0 | sig_meip | reg_seip (or mvip[9]) | reg_seip
1 | 1 | sig_meip | sig_meip | reg_seip

omiga-wbb, 2025-05-15:
What are the differences between the reg_seip of mip[9] and the reg_seip of mvip[9] in the table?