riscv/riscv-aia #67: what's the purpose of

zhuotianshu, 2024-02-25:


zhuotianshu, 2024-02-27:
Now I get that virtual interrupt is for hypervisor to filer the interrupt directed to vs mode, but I still don't have an idea about the concrete scenario in which this feature may be useful.

lixiaogangkkk, 2024-08-14:
@zhuotianshu 
You should read more about virtualization, to get the detail about para-virtualization and full-virtualization
1. Virtual interrupts mechanism is designed to direct interrupt into virtual supervisor (most of time, it's a guest OS running at VS privilege level, such as Linux Kernel, Windows Kernel).
2. Most of the time, if hardware doesn't support interrupts to be directed into virtual supervisor, the host OS running at HS privilege level should take over interrupts which the VM cares about, then host OS emulates interrupt to notify guest OS to handle its interrupt one step further.

lixiaogangkkk, 2024-08-14:
> Now I get that virtual interrupt is for hypervisor to filer the interrupt directed to vs mode, but I still don't have an idea about the concrete scenario in which this feature may be useful.

@zhuotianshu 
You should read more about virtualization, to get the detail about para-virtualization and full-virtualization
1. Virtual interrupts mechanism is designed to direct interrupt into virtual supervisor (most of time, it's a guest OS running at VS privilege level, such as Linux Kernel, Windows Kernel).
2. Most of the time, if hardware doesn't support interrupts to be directed into virtual supervisor, the host OS running at HS privilege level should take over interrupts which the VM cares about, then host OS emulates interrupt to notify guest OS to handle its interrupt one step further.

Steven-Li-Xiaogang, 2025-01-04:
> Thank you for replying @lixiaogangkkk . I finally understood this, but now I'm confused about what should be the priority of virtual interrupts, such as a SEI directed to S mode generated by setting `mvip.SEI`.

while injecting SEI interrupt, it won't try to change the default priority of priority setting from indirection iprio2 (SIE 9). 
if mideleg.SEI=1, writing mvip.SEI and do 'sret' will cause trap into S-mode if sie.SIE is set, sip is alias to mip at this time
if mideleg.SEI=0 and mvien.SEI=1, writing mvip.SEI and do 'sret' will cause SEI trap into S-mode if sie.SIE is set, sip is alias to mvip at this time

the AIA spec also mentioned, when mvien.SEI = 0 mvip.SEIP is alias to mip.SEIP