riscv/riscv-aia #71: Are priorities of virtual interrupts configurable to HS-mode through iprio array?

YenHaoChen, 2024-03-11:
I want to clarify whether priorities of the virtual interrupts, i.e., VSEI, VSTI, and VSSI, are configurable to HS-mode through iprio array if supporting H-extension. The spec has some words about setting a priority to the VSEI through the iprio array (bits 23:16 of iprio2). Thus, I assume the answer is yes, and the HS-mode may configure the virtual interrupts, including VSEI, though the iprio array if supporting H-extension.

However, I was confused by another sentence: ```For a given interrupt number, if the corresponding bit in SIE is read-only zero, then the interrupt’s priority number in the supervisor-level iprio array must be read-only zero as well.``` All virtual interrupt bits in SIE are read-only 0, including SIE.VSEI. According to the above sentence, the corresponding bits of the iprio array (bits 23:16 of iprio2) must be read-only zero, which I believe wasn't the intended conclusion from the spec.

Does the following sentence provide a better description when supporting H-extension? ```If an interrupt number corresponds to a read-only zero bit in both SIE and HIE (if supported), then the interrupt’s priority number in the supervisor-level iprio array must also be read-only zero.```

[image]

YenHaoChen, 2024-03-13:
@jhauser-us Would you share some words about this? Thank you.

jhauser-us, 2024-03-26:
> Does the following sentence provide a better description when supporting H-extension? `If an interrupt number corresponds to a read-only zero bit in both SIE and HIE (if supported), then the interrupt’s priority number in the supervisor-level iprio array must also be read-only zero.`

Yes.  The current text overlooks the writable bits in `hie`.

YenHaoChen, 2024-03-26:
@jhauser-us Thank you. shall I provide a PR of the sentence before closing this issue?

> Does the following sentence provide a better description when supporting H-extension? `If an interrupt number corresponds to a read-only zero bit in both SIE and HIE (if supported), then the interrupt’s priority number in the supervisor-level iprio array must also be read-only zero.`