riscv/riscv-aia #73: Does SEI in VS mode use default priority?

zhuotianshu, 2024-03-12:
On page 74:

> Major interrupts outside the priority-confi gurable subset can still be directed to VS level, but their priority will simply be the default order defi ned in Section 5.1.

But SEI is also not in the subset of major interrupt numbers listed on page 73:

> 1 Supervisor software interrupt
> 5 Supervisor timer interrupt
> 13 Counter overfl ow interrupt
> 14–23 Reserved for standard local interrupts

So I wonder whether SEI in VS mode uses default priority.

vithursons, 2024-03-24:
The statement on page 74 about major interrupts outside the priority-configurable subset using the default priority order is a part of scenario 4 in Section 6.3.3, where hvictl.VTI = 0. However, the priority of a VS-level supervisor external interrupt (SEI) is determined by the first three scenarios in Section 6.3.3. Therefore, the priority of an SEI in VS-mode is not always based on the default priority order, as it depends on the specific conditions outlined in scenarios 1, 2, and 3 of Section 6.3.3.

jhauser-us, 2024-03-26:
Yes, follow the rules in Section 6.3.3.  Comments like the one quoted at the top (about the "major interrupts outside the priority-configurable subset") are intended to be helpful, but the *normative* text (not in comments) has priority.

Modifying the quoted comment to account for SEI would make it more complicated, which may be worse for understanding.  I'm not sure I know how to make it better.