riscv/riscv-aia #74: Why the priority of the third candidate of vstopi is 256?

zhuotianshu, 2024-03-12:
On page 76:

> • if bit 9 is one in both vsip and vsie, and neither of the fi rst two cases applies:
> + a supervisor external interrupt (code 9) with priority number 256;

Since on page 77:

> If bit IPRIOM (IPRIO Mode) of hvictl is zero, IPRIO in vstopi is 1; else, if the priority number for the highest-priority candidate is within the range 1 to 255, IPRIO is that value; else, IPRIO is set to either 0 or 255 in the manner documented for stopi in Section 5.4.2.

It seems that the priority will be regulated to 255, so I wonder why use 256 rather than use 255.

vithursons, 2024-03-24:
[image]
[image]

When scenerios 3 and 4 in section 6.3.3 occur together.

```
1. When DPR = 1, the regular VS-level external interrupt (scenario 3) with IPRIO = 256 will take precedence over the VTI interrupt (scenario 4), regardless of the hvictl.IPRIO value, because 256 is greater than the maximum allowed value of 255 for hvictl.IPRIO.
2. When DPR = 0 and hvictl.IPRIO = 0, the VTI interrupt (scenario 4) will take precedence over the regular VS-level external interrupt (scenario 3) with IPRIO = 256, because an IPRIO of 0 represents the highest priority.
3. When DPR = 0 and hvictl.IPRIO is non-zero, the VTI interrupt (scenario 4) will take precedence over the regular VS-level external interrupt (scenario 3) with IPRIO = 256, because any non-zero value of hvictl.IPRIO (1-255) is considered to have higher priority than an IPRIO of 256.
4. When updating vstopi, if the highest-priority interrupt is the regular VS-level external interrupt with IPRIO = 256, the vstopi.IPRIO field will be set to 255, keeping the value within the valid range of 0-255 while maintaining the correct relative priority order.
```

I think, the choice of 256 as the priority number for scenario 3 is to ensure specific ordering as explained above when scenario 3 and 4 occurs together.

zhuotianshu, 2024-03-25:
Hi, @vithursons. You said:

> 1. When DPR = 1, the regular VS-level external interrupt (scenario 3) with IPRIO = 256 will take precedence over the VTI interrupt (scenario 5), regardless of the hvictl.IPRIO value, because 256 is greater than the maximum allowed value of 255 for hvictl.IPRIO.

In this senario, VSEI has lower priority than VTI interrupt. So in my opinion, VTI interrupt should take precedence over VSEI instead, and DPR is only for cases that VSEI and VTI interrupt have same priority?

vithursons, 2024-03-25:
> In this scenario, VSEI has lower priority than VTI interrupt. So in my opinion, VTI interrupt should take precedence over VSEI instead, and DPR is only for cases that VSEI and VTI interrupt have same priority?

Yeah you are correct I corrected the comment accordingly

vithursons, 2024-03-25:
>I'm not clear about when scenario 3 will happen. Does it happen when set VSEI with hvip
yeah and also when first two scenarios doesn't apply.

jhauser-us, 2024-03-26:
Among the five possible interrupt candidates listed in Section 6.3.3 ("Virtual supervisor top interrupt CSR (`vstopi`)"), items 3 and 4 may occur together, or items 3 and 5 may occur together.  After an implementation determines which interrupt candidates satisfy the listed conditions (never more than two), it must select "the highest-priority interrupt from among the candidates".  The reason item 3 has priority number 256 is so all other candidates have higher priority *except* those with priority number zero and default priority below supervisor external interrupts.

Imagine that item 3 were specified with priority number 255, not 256, and consider the case that `hvictl`.VTI = 0 and bit 13 = 1 in both `vsip` and `vsie`, indicating a local counter overflow interrupt (LCOFI).  Let's assume also that the priority number for LCOFI in `hviprio1` equals 255.  The counter overflow interrupt would satisfy item 4, and in this hypothetical, it would be judged to have lower priority than item 3, because both items 3 and 4 have priority number 255, and the tie-breaker is the default priority order.  If you check above, you'll see that's different from what I said we want.  We want any other candidate to be higher priority than item 3 *unless* that other candidate's priority number is set to zero and its default priority is below SEI.  Here, the priority number for the LCOFI (item 4) is 255, not zero, yet it is being evaluated as lower priority than the SEI (item 3).  Since that's not what we want, specifying item 3 with priority number 255 was a mistake; it needs to have a priority number larger than 255 to always get the desired priority ordering.

The desired ordering can be achieved by giving item 3 any priority number 256 or greater, so it is specified as 256.

zhuotianshu, 2024-03-26:
@jhauser-us Thank you very much, I'll close this issue.

closed 2024-03-26 by zhuotianshu