riscv/riscv-aia #84: Clarify "Access control by the state-enable CSRs"

wissygh, 2024-06-12:
Hi~, when reading about the feature in the spec, I am kind of confused. Could you please clarify the specification for the following feature(in chapter 2.5)：
  Can bit 59 in stateen for riscv64 control access to states of mip/mie?
[image]

jhauser-us, 2024-06-16:
@wissygh:  No bits in `mstateen0` are supposed to affect accesses to M-level CSRs such as `mip` and `mie`.  Hence, I don't understand the question:
> Can bit 59 in stateen for riscv64 control access to states of mip/mie?

The answer would seem to obviously be "No".

wissygh, 2024-06-16:
> @wissygh: No bits in `mstateen0` are supposed to affect accesses to M-level CSRs such as `mip` and `mie`. Hence, I don't understand the question:
> 
> > Can bit 59 in stateen for riscv64 control access to states of mip/mie?
> 
> The answer would seem to obviously be "No".

Sorry, I made a typo. It should be "sip/sie". It only mentions "sieh/siph" here.

closed 2024-06-17 by wissygh