riscv/riscv-aia #91: What is

zhuotianshu, 2024-08-12:


zhuotianshu, 2024-10-11:
It seems that ARC has noticed this problem (https://lists.riscv.org/g/tech-privileged/topic/arc_minutes_2024_09_03/108421694).

jhauser-us, 2024-10-12:
@zhuotianshu, the ARC noticed this problem because you reported it.  Thank you for doing so.

Unfortunately, updating the AIA specification has been blocked until the conversion to AsciiDoc format is complete (which hopefully won't be too much longer).

The ARC decided the answer to your question will be your third suggestion, matching the behavior in VS mode:  If `sip` bit 9 (SEI) is one but there is no interrupt from the external interrupt controller, then the faux SEI has a priority number of 256.  This affects `stopi` the same as if the external interrupt controller were signaling an interrupt with priority 256.

YenHaoChen, 2024-10-18:
@zhuotianshu @jhauser-us 

This discussion clarifies that the SEI from the mvien/mvip has a priority of 256. 

How about the priority of the SEI from PLIC or APLIC, i.e., eidelivery=0x40000000? The figure below in the spec seems to allow any priority between 1~255. However, according to my reading, no specific wording exists about the SEI priority from PLIC or APLIC. I am wondering whether there is a recommended priority for PLIC or APLIC, e.g., 1 or 255? Or, does the AIA spec only restrict the priority not to be 0 or larger than (equal to) 256, and it leaves the priority to be implementation-defined?

[image]

jhauser-us, 2024-10-30:
> How about the priority of the SEI from PLIC or APLIC, i.e., eidelivery=0x40000000?

A PLIC or APLIC assigns priority numbers to external interrupt minor identities.  For an APLIC, the priority of each interrupt source is configured in the source's corresponding `target` register, in field IPRIO.  In the table you copied, where it says "Priority number from interrupt controller (APLIC or IMSIC)", it means the priority number *comes from* the APLIC or IMSIC.

If it is not practical to transmit a signaled interrupt's priority number from a PLIC/APLIC to the RISC-V hart, then you cannot support intermixing priorities of external interrupts with other major interrupts.  In that case, your only option is to implement the `iprio` arrays entirely as read-only zeros.

YenHaoChen, 2024-10-31:
@jhauser-us Thank you for the detail reply. Your reasoning makes sense to me.