riscv/riscv-aia #93: AIA: contradiction in bits 12:0 of hvip description

evgeniy-paltsev, 2024-09-03:
The current AIA documentation has a contradiction in a part describing bits 12:0 of `hvip`:

> Bits 12:0 of `hvip` are reserved and must be read-only zeros, while bits 12:0 of `hvip` are defined by the RISC-V Privileged Architecture. Specifically, bits 10, 6, and 2 of `hvip` are writable bits...

We specify (without any condition) that 12:0 bits of hvip are read-only zeros and writable at a same time.
It seems to be a typo and the first part of phrase could be about `hvien` instead of `hvip`. So, proper phrasing could be:

> Bits 12:0 of **_hvien_** are reserved and must be read-only zeros

Could you please confirm if it is just a typo or I misunderstand the phrasing in the docs?

jhauser-us, 2024-09-10:
The document you are looking at is the one generated from the AsciiDoc sources, which still contain many errors.  In my opinion, it should never have been released as the "correct" version, because it's often not correct.  I just deleted that ill-advised release.

Please discard the file you have.  You can obtain the specification that was actually ratified here:
https://github.com/riscv/riscv-aia/releases/tag/1.0

Eventually, the AsciiDoc version will be made to match the ratified specification, but it's not there yet.

evgeniy-paltsev, 2024-10-09:
Marking as closed as the problematic release was removed.

closed 2024-10-09 by evgeniy-paltsev