riscv/riscv-aia #94: Can AIA only support 2047 EIID maximum?

zhangdujiao, 2024-09-05:
The IMSIC interrupt file supports 2047 interrupts maximum, that is, the range of External Interrupt Identity (EIID) is 1~2047. What should be done if more interrupt IDs are required? 
The EIID configured in the APLIC target register is 10 bits wide and cannot represent more IDs.

avpatel, 2024-09-05:
We have one IMSIC next to each HART.

The Linux IMSIC driver treats M interrupt IDs on each HART as separate MSI vector so for N HARTs we have NxM MSI vectors. In other words, for 10 HARTs with 1023 interrupt IDs per-HART, we will get 10230 MSI vectors. We have sufficiently interrupt ID space so no need to go beyond 2047 interrupt IDs in HW.

zhangdujiao, 2024-09-05:
> The Linux IMSIC driver treats M interrupt IDs on each HART as separate MSI vector so for N HARTs we have NxM MSI vectors

Thanks, Does this mean that the interrupt service routines of different cores should be different, so that different interrupt processing can be performed for the same EIID?

avpatel, 2024-09-05:
Existing interrupt service routines of various Linux drivers work fine without any change.

closed 2024-11-16 by zhangdujiao