riscv/riscv-aia #98: Can hvictl inject an interrupt with IID=0 and IPRIO=0?

zhuotianshu, 2024-10-11:
Spec explicitly states that such an interrupt cannot be generated in M mode:

> To ensure that` mtopi` is never zero when an interrupt is pending and enabled for machine level, if major interrupt 0 can trap to M-mode, it must have a default priority lower than a machine external interrupt. 

And so do S mode. But there seems no such statement around the description of `vstopi`. So what will happen if `hvictl` has `VTI`=1, `IID`=0 and `IPRIO`=0?

jhauser-us, 2024-10-13:
I assume you meant also to add the conditions that `hvictl` has DPR = 0 and IPRIOM = 1.  In that case, the answer to the title's question is, you can try to inject that interrupt, but it won't work.

With all fields of `hvictl` being zeros except VTI = 1 and IPRIOM = 1, it is possible for the interrupt represented by `hvictl` to be the highest-priority candidate for `vstopi`, which will leave `vstopi` with IID = 0 and IPRIO = 0.  But then no interrupt will actually be taken, because `vstopi` is all zeros.

If any other lower-priority interrupts are pending-and-enabled, this value of `hvictl` will actually block them from being taken.

Note that if `hvictl`.IPRIOM = 0, the value of `vstopi`.IPRIO will be 1.  And if `hvictl` has IPRIOM = 1 and DPR = 1 (and if there is no pending-and-enabled external interrupt), then `vstopi`.IPRIO will be 255.  Those settings will allow an interrupt with IID = 0 to occur.

jhauser-us, 2025-02-14:
> 1. Would hvictl (even VTI=0) also affect the pending CSR sip(actually vsip)? if yes, sip is logical-OR hvip and external interrupt controller with hvictl, or any other way? else not, it only takes effects on interrupt taken flow?

The value of `hvictl` affects no other registers except `vstopi`.  An interrupt trap is taken in VS-mode when `vstopi` is not zero and `vsstatus`.SIE = 1.

> 2. if hvictl takes effect on interrupt taken flow, when VTI=0, should the interrupt enable bit hvictl.IID be set at the same time?

The value of `hvictl` is changed only when software explicitly writes this CSR; it never changes automatically.

> 3. if hvictl takes effect on interrupt taken, how the interrupt injected by hvictl cleared, is it cleared by reading vstopi automatically when hvictl.IID is the highest priority? or any other way?

See the answer to question 4 in issue #115.

jhauser-us, 2025-08-12:
> Does this mean that the vstopi.IPRIO must be 1, no matter which candidate is ultimately selected?

Yes.  When `hvictl`.IPRIOM = 0 and `vstopi` is not zero, then the IPRIO field of `vstopi` = 1.

senlinhe, 2025-08-13:
so we can say hvictl.IPRIOM=0 does not affect the candidates' IPRIO for arbitration, but it only effect the final outcome reflected in vstopi? 
Why do we need such mechanism?

jhauser-us, 2025-08-13:
> so we can say hvictl.IPRIOM=0 does not affect the candidates' IPRIO for arbitration, but it only effect the final outcome reflected in vstopi?

Correct.

> Why do we need such mechanism?

Because of this in Section 5.4.2, "Supervisor top interrupt CSR (`stopi`)":
> If all bytes of the supervisor-level `iprio` array are read-only zeros, a simplified implementation of field IPRIO is allowed in which its value is always 1 whenever `stopi` is not zero.

Steven-Li-Xiaogang, 2025-12-24:
@jhauser-us 
> Note that if `hvictl`.IPRIOM = 0, the value of `vstopi`.IPRIO will be 1. And if `hvictl` has IPRIOM = 1 and DPR = 1 (and if there is no pending-and-enabled external interrupt), then `vstopi`.IPRIO will be 255. Those settings will allow an interrupt with IID = 0 to occur.

When DPR=1, according to AIA spec table 5.5, the lowest iprio should be 0, then the the vstopi.IPRIO still will be 0, i think there still won't allow an interrupt with IID=0  occur.

jhauser-us, 2026-01-12:
> When DPR=1, according to AIA spec table 5.5, the lowest iprio should be 0, then the the vstopi.IPRIO still will be 0,

You're missing a step.  The end of Section 6.3.3, "Virtual supervisor top interrupt CSR (`vstopi`)", says that if IPRIOM = 1 and `hvictl`.IPRIO is not within the range 1 to 255, then `vstopi`.IPRIO "is set to either 0 or 255 in the manner documented for `stopi` in Section 5.4.2."  Looking to Section 5.4.2, "Supervisor top interrupt CSR (`stopi`)", it says, "If the interrupt’s priority number is zero and interrupt number IID has a default priority lower than a supervisor external interrupt, then IPRIO is 255 (lowest representable priority)."  This combination gives what I wrote before:  "if `hvictl` has IPRIOM = 1 and DPR = 1 (and if there is no pending-and-enabled external interrupt), then `vstopi`.IPRIO will be 255."

When `hvictl`.IPRIOM = 1, the value of `vstopi`.IPRIO is usually just a copy of `hvictl`.IPRIO, but this is a case when it is not.

> By the way, as IID field is WARL, could implementation select to treat interrupt id 0 as an illegal or unimplemented value and do not support inject interrupt id 0?

From Section 6.3.2, "Virtual interrupts for VS level", which describes `hvictl`:  "If *k* bits are implemented for IID, then all values 0 through 2^*k* - 1 are supported."  So you must support interrupt identifier 0 for `hvictl`.IID.

Steven-Li-Xiaogang, 2026-01-12:
@jhauser-us thanks a lot for your clarification