riscv/riscv-aia #100: Limitation access to the M-mode domain APLIC without PMP entries

nonufrienko, 2024-10-15:
Can an implementation without PMP restrict access to the APLIC machine-level interrupt domain memory region from a less privileged mode?

Is it acceptable to always report an access fault in such cases?

jhauser-us, 2024-11-17:
Your question is not really about an APLIC but instead is this:  Can a RISC-V machine implement some form of privilege-specific physical memory protection that is not the PMP specified by the RISC-V Privileged ISA?

I don't have an official answer.  I suggest asking in a forum where the Privileged ISA is discussed.  If you're a member of RISC-V International, you could E-mail the 'tech-privileged' mailing list.  If you're not a member, maybe here:
https://groups.google.com/a/groups.riscv.org/g/isa-dev

nonufrienko, 2024-11-20:
Ok, thanks

closed 2024-11-20 by nonufrienko