riscv/riscv-aia #99: Is hstatus.VGEIN used to index guest interupt files when accessing iCSR via VSISELECT/VSIREG window?

zengderui, 2024-10-15:
If I understand the spec correctly, each guest interrupt file has it own iCSR register file, and hstatus.VGEIN selects the iCSR file when VS-level iCSR files. E.g. when hstatus.VGEIN is 3 and VSISELECT is 0x70, accessing VSIREG ends up accessing eideilivery iCSR of guest interrupt file 3.

It seems the spec does not **explicitly** state how guest interrupt file is selected, I'd appreciate it if anyone clarifies it. Thanks in advance!

jhauser-us, 2024-11-17:
In the IMSIC chapter, in the subsection titled "CSRs for external interrupts via an IMSIC", the AIA document says:
> When an IMSIC has guest interrupt files, the VS CSRs interact with a single guest interrupt file, selected by the VGEIN field of CSR `hstatus`.

> ... when the hypervisor extension is implemented, there are three corresponding VS CSRs: `vsiselect`, `vsireg`, and `vstopei`.

> ... In each case, a value of the *iselect CSR (`miselect`, `siselect`, or `vsiselect`) in the range 0x70–0xFF selects a register of the corresponding IMSIC interrupt file, either the machine-level interrupt file (`miselect`), the supervisor-level interrupt file (`siselect`), or a guest interrupt file (`vsiselect`).

Maybe it's not easy to find, but I do think the AIA spec says `hstatus`.VGEIN selects the guest interrupt file that is accessed through registers `vsiselect`, `vsireg`, and `vstopei`.

zengderui, 2024-11-18:
Thanks for clarifing!

closed 2024-11-18 by zengderui