riscv/riscv-aia #102: Supervisor-level iprio configurability

evgeniy-paltsev, 2024-10-31:
The AIA spec says
> For a given interrupt number, **if the corresponding bit in sie is read-only zero**, then the interrupt’s priority number in the supervisor-level iprio array must be read-only zero as well.

Does it include the case when `sie[n]` is read-only zero because corresponding `mideleg[n]` & `mvien[n]` bits are both set to zero?
I'm assuming here that either `mideleg[n]` or `mvien[n]` (or both) is writable and can be set to 1.

[image]

jhauser-us, 2024-11-17:
First, I should add a correction:  When the H (hypervisor) extension is implemented, the AIA spec should say, "if the corresponding bit is read-only zero **in both `sie` and `hie`**, then the interrupt’s priority number in the supervisor-level `iprio` array must be read-only zero as well."

> Does it include the case when `sie[n]` is read-only zero because corresponding `mideleg[n]` & `mvien[n]` bits are both set to zero?

Yes, it does.

jhauser-us, 2024-12-12:
> I'm wondering if there is any requirements for s-mode `iprio[n]` value when `iprio[n]` field stops being read-only zero after we write 1 to corresponding `mideleg[n]` or `mvien[n]`?

This question is answered by the Privileged ISA spec, in the section titled "CSR Field Modulation":
> If a write to one CSR changes the set of legal values allowed for a field of a second CSR, then unless specified otherwise, the second CSR’s field immediately gets an UNSPECIFIED value from among its new legal values.