riscv/riscv-aia #103: When vsiselect is in the range of 0x070–0x0FF, does access to vsireg2 belong to IMSIC states?

Jahhow, 2024-11-05:
If the access belong to IMSIC states, then the access can be controled by stateen[58] (IMSIC),
so when mstateen0[58]=0, the access to vsireg2 shall raise illegal-instruction-exception.

Otherwise, if the access does not belong to IMSIC states, then the access should follow this rule in the AIA specification:
> when vsiselect has a value **in the range 0x30–0x3F or 0x70–0xFF**, attempts from M-mode or HS-mode to access alias CSRs vsireg2 through vsireg6 raise an illegal instruction exception, and attempts **from VS-mode to access sireg2 through sireg6 raise a virtual instruction exception**.

jhauser-us, 2024-12-12:
For the AIA as it currently exists, the answer to the title's question is "no".  The sentence that you quoted is not overridden by `mstateen0` bit 58 (IMSIC) or bit 59 (AIA), only by `mstateen0` bit 60 (CSRIND).

Jahhow, 2024-12-13:
Thanks for the clarification.

closed 2024-12-13 by Jahhow