riscv/riscv-aia #105: A question of  Interrupt filtering？

ZeyueShen, 2024-11-20:
According to spec 5.3, mvip and mvien can be used to implement interrupt filtering. Here's an idea for an interrupt filtering method: Firstly, an array of external interrupt ids that need to be filtered. Secondly, s mode external interrupt is generated for an id stored in this array in S mode. After entering the trap, read stopei to obtain the id, and then configure mvien[9]=1, mvip[9]=1, mie[9]=0. After exiting the trap, Then enter the virtual interrupt trap. At this time, csrrw stopei will report illegal instrction exception, and then jump to the exception trap in M mode to clear mvien[9], mvip[9] and csrrw stopei. At this time, the information obtained through csrrw stopei will be stored in a total of previously defined public variables. This public variable is then used after exiting the trap so that S mode can obtain information about the external interrupt. Is it possible to pass the obtained external interrupt information to S mode by using public variables?

jhauser-us, 2024-12-12:
I'm not sure how to respond to this issue.
> Secondly, s mode external interrupt is generated for an id stored in this array in S mode. After entering the trap, read stopei to obtain the id, and then configure mvien[9]=1, mvip[9]=1, mie[9]=0.

Since machine-level registers like `mvien`, `mvip`, and `mie` aren't accessible from S mode, I assume you mean that `mideleg`[9] = 0 and the SIE traps to M mode?

> After exiting the trap, Then enter the virtual interrupt trap. At this time, csrrw stopei will report illegal instrction exception, ....

You definitely lose me here, because `stopei` cannot report an illegal instruction exception (or any other synchronous exceptions).

> Is it possible to pass the obtained external interrupt information to S mode by using public variables?

Yes, in theory.  But the problem is that S-level software (usually an operating system) has to be written to know to look in those variables for the information.  Standard operating systems you can download and install on your machine won't normally do that.

ZeyueShen, 2024-12-13:
hi @jhauser-us -Thank you for your answer.
[image]
Yes, It's true. 
[image]
The causes of stopei exceptions are described in spec (The RISC-V Advanced Interrupt Architecture: 5.3):
[image]

[image]
Ok.We are mainly used to verify the feasibility of the hardware, so we need to have a general verification of S-level software behavior, and we are not specialized in S-level software.

jhauser-us, 2024-12-13:
Oh, you meant the attempt to access `stopei` causes an illegal instruction exception.  I understand now.

I guess the sequence you are suggesting would work.  Not sure why you're doing it.  (But I don't need to know.)

More typically, M-level code would emulate the trapped attempt to access `stopei`, storing the desired interrupt information in the CSR instruction's destination register before returning from the trap, so S-level software could then proceed in the usual way.  But there's no ISA rule requiring you to do things that way, if you want to change the S-level software to work differently.

ZeyueShen, 2024-12-13:
[image]
It is mainly based on this paragraph from spec that we envision a way for software to run：
[image]
Thank you for your answer. I understand