riscv/riscv-aia #106: Questions about mvip

Baixyzz, 2024-11-25:
Hi,@jhauser-us.I have a question about the design purpose of mvip.AIA spec says： #The same hardware may also, under the right circumstances, allow machine level to assert virtual interrupts to supervisor level that have no connection to any real interrupt events.
In context, I understand that the main design purpose of mvip is to assert virtual interrupts into supervisor level.In the Privileged ISA,we also can see that mip.SEIP,mip.SSIP and mip.LCOFIP are writable.So I understand M-mode software can use mip to inject interrupts into  supervisor level.In summary, I am confused about what scenarios software must use mvip instead of mip. Or, what things can software do by using mvip but mip cannot?

Baixyzz, 2024-11-29:
@jhauser-us Would you share some words about this? Thank you.

jhauser-us, 2024-12-13:
The main purpose of `mvien` and `mvip` is to support this sequence:
1. A major interrupt traps to M-mode.
2. M-level software decides to pass on this interrupt to S-mode.

Let's assume we want to support this sequence for major interrupt N and we don't have `mvien` and `mvip`.  Bit N of `mideleg` must be either 0 or 1.  If `mideleg`[N] = 1, then we can't do step 1, because interrupt N will be immediately delegated to another privilege mode, not M-mode.  And if `mideleg`[N] = 0, then we can't do step 2, because bit N is forced to be read-only zero in both `sip` and `sie`, according to the existing Privileged ISA spec.

Baixyzz, 2024-12-13:
@jhauser-us Thank you very much, I'll close this issue.

closed 2024-12-13 by Baixyzz