riscv/riscv-aia #107: Can we delivery external interrupt connected to APLIC to an virtual hart whose interrupt file is located in Memory (MRIF)

Steven-Li-Xiaogang, 2024-12-01:
As the AIA mentioned to extend IOMMU with MSI flatten page table to support interrupt virtualization, but it only mentioned that MSI message target to an virtual hart requires to map it's MSI message such as PCIe virtual endpoint device direct to an virtual machine （MRIF or interrupt file) which is using a virtual address to be translated to physical address. 
1. For interrupt file in IMSIC which can be addressed by APLIC, but for MRIF associated virtual hart which does not have a dedicated interrupt file and can not be addressed by APLIC, how could we delivery external interrupts to it?
2. for IOMMU received an MSIs to toward a MRIF, it requires to send a MSI notify to hypervisor, it seems AIA spec and IOMMU spec didn't mention the format of this MSI notify message, does it means it's designed by implementation and not necessary to be exposed to software?

Thanks,
Steven Li

Steven-Li-Xiaogang, 2024-12-10:
@jhauser-us would you share the detail solution to delivery APLIC interrupt to a MRIF banded virtual machine (guest os)?

jhauser-us, 2024-12-12:
We do not expect a guest OS to be given access to configure an APLIC directly.  Even if a guest OS thinks it is accessing an APLIC directly, those attempted accesses must be trapped and emulated by the hypervisor.  Consequently, the hypervisor can easily have complete knowledge of which APLIC interrupts are going to which guest interrupt files.

If an interrupt file is moved ("swapped out") to an MRIF, the hypervisor must also reconfigure the relevant APLIC interrupts to go to itself (the hypervisor).  Whenever it receives such an interrupt, the hypervisor can decide what to do.  As you observed, the AIA does not provide a way for an APLIC to autonomously set an interrupt-pending bit in an MRIF.  This limitation was an intentional choice.

jhauser-us, 2024-12-12:
> 2. for IOMMU received an MSIs to toward a MRIF, it requires to send a MSI notify to hypervisor, it seems AIA spec and IOMMU spec didn't mention the format of this MSI notify message, ....

Yes the AIA spec does.  In the last subsection titled "MSI PTE, MRIF mode", it says:
> Field NPPN (Notice Physical Page Number) and the two NID (Notice Identifier) fields together specify a destination and value for a *notice MSI* that is sent after each time the destination MRIF is updated as a result of consulting this PTE to store an incoming MSI.

and at the very end of the section:
> Once the MRIF update operation is visible to all agents in the system, the 11-bit NID value is zero-extended to 32 bits, and this value is written to the address NPPN << 12 (i.e., physical page number NPPN, page offset zero) in little-endian byte order.

Steven-Li-Xiaogang, 2024-12-19:
> We do not expect a guest OS to be given access to configure an APLIC directly. Even if a guest OS thinks it is accessing an APLIC directly, those attempted accesses must be trapped and emulated by [...]

@jhauser-us if a wired interrupt attached to APLIC directs to a vhart whose interrupt file is swapped out to MRIF, the only solution is to reconfigure the interrupt to hypervisor without delegation temporary. once interrupt happens  hypervisor should swap in the vhart and emulate the interrupt to signal this vhart. the interrupt handling process becomes longer than normal way, would cause performance reduction, seems all interrupts targets to a vhart whose interrupt file was swapped out to MRIF takes more time than the interrupt to vhart which has a dedicated interrupt file.

closed 2025-01-14 by Steven-Li-Xiaogang