riscv/riscv-aia #111: Some question about hvictl

Baixyzz, 2024-12-13:
Hi @jhauser-us,
We have some question about hvictl.
1.From [https://github.com/riscv/riscv-aia/issues/61](url),we can see hypervisor can write hvictl.IID to any value. Even if hvictl.IID is an interrupt number that VS-level cannot support, such as MEI, as long as its priority number is the highest, it will be displayed in vstopi. 
AIA spec say,

>  Instead, execution must resume from a WFI whenever an interrupt is pending at any privilege level.And if the hypervisor extension is
implemented, an interrupt is pending at VS level if vstopi (Section 6.3.3) is not zero.

So even if hvictl injects an interrupt that VS-level does not support, instruction execution must resume from a WFI. Is this correct?

2. If hvictl.IID is an interrupt number that VS-level cannot support, at the same time, the external interrupt controller sends SEI to the virtual hart. But the priority number of SEI is greater than hvictl.IPRIO (hvictl.IPRIOM=1). So vstopi.IID is equal to hvictl.IID. But the core will not respond to the interrupt corresponding to vstopi.IID, so should the core respond to SEI at this time? If not, then this is equivalent to the software using hvictl to prevent the hart from responding to interrupts that have connection to real interrupt events.
3. From AIA 6.3.3 case2, using hvictl to inject SEI to VS level requires ensuring vsip[9]=1. Privileged architecture says，

> When bit 10 of hideleg is zero, vsip.SEIP and vsie.SEIE are read-only zeros. Else, vsip.SEIP and vsie.SEIE are aliases of hip.VSEIP and hie.VSEIE.
Does this mean, hip.VSEIP=1 when hvictl.IID=9?  Furthermore, if bit 10 of hideleg is one, vsip.SEIP is one.

jhauser-us, 2025-02-13:
> 1. ....  So even if hvictl injects an interrupt that VS-level does not support, instruction execution must resume from a WFI. Is this correct?

Yes.

> 2. If hvictl.IID is an interrupt number that VS-level cannot support, at the same time, the external interrupt controller sends SEI to the virtual hart. But the priority number of SEI is greater [...]

When executing in VS-mode, if `vstopi` is not zero and `vsstatus`.SIE = 1, then the hart *will trap*.  The major interrupt number reported in `vscause` will be the value of `vstopi`.IID.  Any belief that this is an interrupt that VS-level "cannot support" is irrelevant.  The interrupt trap *will be taken*.  Your assertion that "the core will not respond to the interrupt corresponding to vstopi.IID" is incorrect.

See also #98.

> 3. From AIA 6.3.3 case2, using hvictl to inject SEI to VS level requires ensuring vsip[9]=1. Privileged architecture says,
> > When bit 10 of hideleg is zero, vsip.SEIP and vsie.SEIE are read-only zeros. Else, vsip.SEIP and vsie.SEIE are aliases of hip.VSEIP and hie.VSEIE.

> Does this mean, hip.VSEIP=1 when hvictl.IID=9? Furthermore, if bit 10 of hideleg is one, vsip.SEIP is one.

No, `hvictl`.IID = 9 does not imply `hip`.VSEIP = 1.  A hypervisor can set bit 10 = 1 in both `hideleg` and `hvip` to set `vsip`.SEIP = 1 and make `vsie`.SEIP writable (not read-only zero).

Baixyzz, 2025-02-13:
Thank you very much for your answer. I will close this issue

closed 2025-02-13 by Baixyzz