riscv/riscv-aia #115: Can hvictl takes effect on pending bits on mip/hip/sip/vsip and impact the interrupt taken without accessing mvip/hvip and without interrupt signals?

Steven-Li-Xiaogang, 2025-01-19:
[image]
 
1. as 6.3.2 of AIA spec mentioned, hypervisor can use hvictl to assert any major interrupts that are not supported by hvip, or inject interrupt with a perferred priority. does it means hvictl can be used standalone and can affect value reading from mip/hip/sip/vsip
2. can we inject an interrupts with major interrupts whose identity belongs to M-mode or S-mode, if yes, does it affect interrupt taken flow if we also specify an user priority?
3. It seems SW also can inject an interrupt with hvictl.VTI=0, except affecting sip/sie/stimecmp accessing in VS-mode, is there any difference from injecting interrupt with hvictl.VTI=1

Thanks,
Steven-Li

Steven-Li-Xiaogang, 2025-02-06:
@gfavor @aswaterman would you help on these questions? thank you very much

aswaterman, 2025-02-06:
Not being the resident AIA expert, I'm hoping someone else can chime in.

gfavor, 2025-02-06:
This should be directed to @jhauser-us to answer reliably.

gfavor, 2025-02-07:
It might also be good to send your questions to the AIA TG email list.

jhauser-us, 2025-02-13:
> 1. as 6.3.2 of AIA spec mentioned, hypervisor can use hvictl to assert any major interrupts that are not supported by hvip, or inject interrupt with a perferred priority. does it means hvictl can be used standalone and can affect value reading from mip/hip/sip/vsip

Register `hvictl` does not affect the value of any other register except `vstopi`.  The value of `hvictl` has no effect on `mip`, `hip`, `sip`, or `vsip`.

> 2. can we inject an interrupts with major interrupts whose identity belongs to M-mode or S-mode, ...?

No, because `hvictl` only affects `vstopi`.

> 3. It seems SW also can inject an interrupt with hvictl.VTI=0, except affecting sip/sie/stimecmp accessing in VS-mode, is there any difference from injecting interrupt with hvictl.VTI=1

Using `hvictl`, the only major interrupt that can be injected when `hvictl`.VTI = 0 is number 9, a supervisor external interrupt (SEI).  But an SEI interrupt is injected only when `vsip`.SEIP and `vsie`.SEIE are both one.

It sounds from your question that you think VS-mode accesses to `sip`, `sie`, and `stimecmp` trap when `hvictl`.VTI = 0.  If so, that is backwards.  The traps occur when `hvictl`.VTI = 1, not 0.

> 4. What's the way to clear interrupt injected by hvictl, should it be cleared once an interrupt is taken or we should re-write hvictl implicitly? or any other way?

The value of `hvictl` is changed only by an explicit write to the register by software.  It is never changed automatically.

For most major interrupts, a VS-level interrupt handler clears the interrupt by zeroing the pending bit in `sip` (really `vsip`).  If the interrupt is not SEI (number 9), then the only way `hvictl` could have injected this interrupt is with `hvictl`.VTI = 1.  Hence, the VS-level interrupt handler's attempt to write to `sip` will cause a virtual instruction exception.  At this point, the hypervisor should interpret the guest's attempt to write to `sip`, and if it clears the interrupt being injected by `hvictl`, then the hypervisor should change the value of `hvictl`.

jhauser-us, 2025-02-13:
> It might also be good to send your questions to the AIA TG email list.

The mailing list is being shut down because the AIA TG no longer exists.

jhauser-us, 2025-02-13:
> 2. can we inject an interrupts with major interrupts whose identity belongs to M-mode or S-mode, if yes, does it affect interrupt taken flow if we also specify an user priority?

I wrote:
> No, because `hvictl` only affects `vstopi`.

You can inject any interrupt number you want to VS level, even ones like 7 (MTI) and 11 (MEI).  What you cannot do is use `hvictl` to cause a trap that is taken in M or HS mode.  Register `hvictl` only affects traps in VS mode.

Steven-Li-Xiaogang, 2025-03-25:
@jhauser-us there is still one question which is not explicit in AIA spec, should hvictl take effect on interrupt taken flow even vsip and vsie are zeros or hardwired to zeros?
If yes, does it mean in VS-mode interrupt taken flow can check CSR vstopi directly instead of checking CSR hideleg, vsie and pending CSR mip/sip/vsip.

Steven-Li-Xiaogang, 2025-03-31:
As software interrupt can be triggered by external interrupt controller (VSEI) or environment (VSTI, LCOFI), how could software distinguish the interrupt source of an interrupt?
1. injecting an MTI/MEI to VS level, it's easy to know the cause is from hvictl because there could not be another way to fire an MTI/MEI to VS level.
2. if an interrupt from external interrupt controller to VS level at the same time with injecting an VSEI by writing hvictl, how could software distinguish the source? or software should not care about the source, just do query/clear the interrupt status from external interrupt controller and also let hypervisor to clear the injected interrupt the same time.
3. if an interrupt is injected by hvip and hvictl at the same time, what about the interrupt processing flow?

jhauser-us, 2025-04-08:
> 2. if an interrupt from external interrupt controller to VS level at the same time with injecting an VSEI by writing hvictl, how could software distinguish the source? or software should not care [...]

Although a VS-level interrupt signal from an external interrupt controller starts off as a VSEI (major interrupt 10), if the interrupt is delegated to VS-level, it appears there as an SEI (major interrupt 9).  In the section titled "Virtual supervisor top interrupt CSR (`vstopi`)", the rules for determining the value of `vstopi` do not permit an SEI to be injected to VS-level by `hvictl` at the same time as a real VSEI is delegated to VS-level as an SEI.

It is possible for `hvictl` to inject a VSEI (major interrupt 10) at VS-level, but a real interrupt from an interrupt controller will never appear as a VSEI at VS-level.

Lastly, the whole idea of a virtual machine is that guest software at VS-level is not supposed to know whether an SEI it sees is a real interrupt or a virtual one.  Ordinarily, the guest software should act as though the SEI is a real interrupt.  If the SEI is actually an injected virtual interrupt, the hypervisor software may have a role in making the SEI appear to the guest as a real interrupt.

> 3. if an interrupt is injected by hvip and hvictl at the same time, what about the interrupt processing flow?

Step 1:  The values of `hvien` and `hvip` affect `vsip`.  Step 2:  The rules in the section titled "Virtual supervisor top interrupt CSR (`vstopi`)" specify how the values of `vsip`, `hvictl`, and other registers affect `vstopi`.  And Step 3:  The value of `vstopi` determines what interrupt, if any, is taken in VS-mode.

Crucially, the AIA section titled "Interrupt traps to VS-mode" begins:
> The Advanced Interrupt Architecture modifies the H extension such that an interrupt is pending at VS level if and only if `vstopi` is not zero. CSRs `vsip` and `vsie` do not by themselves determine whether a VS-level interrupt is pending, though they may do so indirectly through their effect on `vstopi`.

gfavor, 2025-04-12:
Closing since there has been no further questions.

closed 2025-04-12 by gfavor