riscv/riscv-aia #118: SEI (major 9, delegated to S mode) or STI (major 5, not delegated)，which has a higher priority?

ic-nerd, 2025-02-18:
Consider the following scenario: 
1. mideleg.SEIP = 1, SEI (major 9) is delegated to S-mode; 
2. mideleg.STIP = 0, STI (major 5) is not delegated to S-mode and is an M-mode interrupt; 
3. Currently running in S-mode, both SEI and STI are enabled, and sstatus.SIE = 1; 
4. At this point, both SEI and STI interrupts are set to 1. Which interrupt should be responded to first? Which interrupt has a higher priority?

jhauser-us, 2025-02-27:
The RISC-V Privileged ISA document, in the section titled "Machine Interrupt (`mip` and `mie`) Registers", says, "Interrupts to M-mode take priority over any interrupts to lower privilege modes."

jhauser-us, 2025-03-20:
The AIA document has been updated to say:
> The default priority order in Table 8 is applicable only when multiple major interrupts would trap to the same privilege mode. Interrupt traps to a more-privileged mode always have priority over traps to a less-privileged mode.

closed 2025-06-04 by ic-nerd