riscv/riscv-aia #121: MVIP state

kdockser, 2025-03-20:
Section **5.3 Interrupt filtering and virtual interrupts for supervisor level** states:

`When supervisor mode is implemented, the minimal required implementation of mvien and mvip has all bits being read-only zeros except for mvip bits 1 and 9, and sometimes bit 5, each of which is an alias of an existing writable bit in mip.`

While other text does limit what other bits of `mvip` can be implemented, it doesn't explicitly state their connections to other registers.

1. Is it the intent that all implemented (and not reserved) `mvip[n]` registers are aliases of `mip[n]` when `mvien[n]` is 0,  and are aliases of `sip[n]` when `mvien[n]` is 1 (with the following exceptions)?
2. When n=1, it is stated that when `mvien[n]` is one, `mvip[n]` is a seperate writable bit independent of `mip[n]` and this bit becomes UNSPECIFIED when `mvien[n]` is changed from zero to one. Is this _only_ true for n=1, or is it also true for any other values of n?
3. When n=9, it is stated that when mvien[n] is zero, mvip[n] is an alias of the software-writable bit 9 of mip (SEIP). But when mvien[n] is one, mvip[n]=_x_ , where _x_ is a writable bit independent of all components of mip[9] (i.e., independent of the software-writable portion and the interrupt signal from supervisor-level interrupt file). Unlike for n=1, changing the value mvien[9] does not affect the value of _x_. This implies that the value of _x_ persists and is accessible again when mvien[n] is one again. Why does this value persist for n=9 but becomes UNSPECIFIED when n=1?
4. Is it true that when vien[9] is one, mip[9] is only a writable bit, which is the same bit as mvip[9] when mvien[9] is zero?

Thanks,
Ken

jhauser-us, 2025-03-20:
> 1. Is it the intent that all implemented (and not reserved) `mvip[n]` registers are aliases of `mip[n]` when `mvien[n]` is 0, ....

No.  Bits 1 and 9 of `mvip` are specified specially, different than bits 13 and higher.

> 2. When n=1, it is stated that when `mvien[n]` is one, `mvip[n]` is a seperate writable bit independent of `mip[n]` and this bit becomes UNSPECIFIED when `mvien[n]` is changed from zero to one. Is this *only* true for n=1, or is it also true for any other values of n?

It's true only for n = 1 (SSI).

> 3. When n=9, it is stated that when mvien[n] is zero, mvip[n] is an alias of the software-writable bit 9 of mip (SEIP). But when mvien[n] is one, mvip[n]=*x* , where *x* is a writable bit [...]

Because only for `mip` bit 9 (SEIP) does the Privileged ISA specify an unusual, hidden "software-writable bit".  There is not an equivalent hidden bit for `mip` bit 1 (SSIP).

> 4. Is it true that when vien[9] is one, mip[9] is only a writable bit, which is the same bit as mvip[9] when mvien[9] is zero?

I see a couple of errors here.  When `mvien`[9] = 1, `mip`[9] is a *read-only* bit, not writable.  And when `mvien`[9] = 0, `mvip`[9] is the same bit as the Pivileged ISA's hidden *software-writable SEIP bit* that underlies `mip`[9].  The intent, in fact, is that `mvip`[9] is always the same bit regardless of the value of `mvien`[9].

The AIA document was recently updated with this expanded comment:

> The base Privileged Architecture specifies unusual read/write behavior for what it calls the software-writable SEIP bit of register `mip`. When bit 9 of `mvien` is zero, bit 9 of `mvip` makes the software-writable SEIP bit of `mip` directly accessible by itself. Furthermore, as explained below [in the document], setting bit 9 of `mvien` to one separates the software-writable SEIP bit from `mip` entirely, so it is then just a writable bit in `mvip`.

See also issues #64 and #101.

kdockser, 2025-03-20:
@jhauser-us Thanks for your prompt and detailed reply. That helps a lot.

The behavior of the other bits is still unclear to me:

1. For n=5, I see that mvip[5] is an alias for mip[5] when mip[5] is writable and zero otherwise. Is this independent of the value in  mvein[5]? 
2. For n=13 (for example), what is the value of mvip[n] for mvein[n]=0 and mvein[n]=1?

jhauser-us, 2025-03-21:
> 1. For n=5, I see that mvip[5] is an alias for mip[5] when mip[5] is writable and zero otherwise. Is this independent of the value in mvein[5]?

Other than bits 1 and 9, the AIA document explicitly says, "The other bits of `mvien` for interrupts 0-12 are reserved and must be read-only zeros."  So `mvien`[5] is always zero.

> 2. For n=13 (for example), what is the value of mvip[n] for mvein[n]=0 and mvein[n]=1?

For *n* >= 13, assuming bit `mvip`[*n*] is not read-only zero, then it is a writable bit that can be whatever value software writes to it, independent of `mvien`[*n*].

closed 2025-03-21 by kdockser