riscv/riscv-aia #122: vstopi with smstateen

jiahzhang, 2025-04-15:
The value of vstopi depends on vstopei for "scenario 1" of its possible values. If bit 58 of hstateen0 is disabled, I believe this means we should not consider scenario 1 of possible vstopi values.

```
Bit 58 is implemented in hstateen0 only if the hart has an IMSIC. Furthermore, even with an IMSIC, bit 58 may (or may not) be read-only zero in hstateen0 if the IMSIC has no guest interrupt files for guest external interrupts (Chapter 3). When this bit is zero (whether read-only zero or set to zero), a virtual machine is prevented from accessing the hart’s IMSIC the same as when hstatus.VGEIN = 0.
```

If this is the case, does this mean potentially an M-mode read of vstopi (where presumably vstopei is accessible) would be different from a VS-mode read of vstopi?

radimkrcmar, 2025-04-17:
The hstateen0 bits only change the permissions of VS/VU mode CSR accesses done by the CSR instructions -- other aspects of the registers are not affected.  The hardware vstopi state is used in VS/VU mode interrupt delivery.

jiahzhang, 2025-04-17:
But accessing vstopi is a CSR instruction right?

radimkrcmar, 2025-04-17:
Is the following correctly describing your case?

* The hart has an IMSIC and hstatus.VGEIN is set to an IMSIC whose vstopei is non-zero.
* hstateen0 bit 58 is clear and the others are set.
* VS mode executes the "read stopi CSR", which is effectively accessing vstopi, or M-mode executes "read vstopi CSR".
* Neither of the accesses trap and the CSR value is provided by hardware.
* The value of vstopi depends on vstopei, but VS mode behaves as of hstatus.VGEIN was zero.
* VS and M read different CSR values.

I interpret the hstateen0 bit 58 differently:
The CSR reads will return the same value, because only explicit (CSR instructions) VS accesses to vstopei are prevented.
The indirect effect of vstopei on the VS visible stopi value is allowed.
VS will think that it doesn't have an IMSIC and the interrupts are getting delivered with a different mechanism, but the underlying interrupt delivery mechanism doesn't consider hstateen0.

Could that work?

jiahzhang, 2025-04-17:
Perhaps, but even for example bit 58 we protect against the IMSIC state through the *iselect, *ireg CSRs. This would be an indirect access (not protecting the CSR itself).

If it's as you say, then yes, M/VS would observe the same value through vstopi. Although it begs the question as to how this interrupt would get cleared.

radimkrcmar, 2025-04-17:
> Perhaps, but even for example bit 58 we protect against the IMSIC state through the *iselect, *ireg CSRs. This would be an indirect access (not protecting the CSR itself).

HS mode must have explicitly configured hstatus.VGEIN to activate the IMSIC, which I take as a consent with the information leakage.

> If it's as you say, then yes, M/VS would observe the same value through vstopi. Although it begs the question as to how this interrupt would get cleared

That is HS mode's problem. :)

jiahzhang, 2025-04-17:
Ok thanks

closed 2025-04-17 by jiahzhang