riscv/riscv-aia #123: Question about the mapping between Child index and the domain index

Yasinhwang, 2025-04-20:
AIA spec chapter 4.5.2 says:

_"Child Index is a WLRL field that specifies the interrupt domain to which this source is delegated.
 For an interrupt domain with C child domains, this field must be able to hold integer values in the
 range 0 to C −1. Each interrupt domain has a fixed mapping from these index numbers to child
 domains"_

Hardware design needs to use the mapping to delegate the interrupts to child domains from parent domain, the mapping relationship should be knew by hardware. Then how to find the mapping betwen child index and the real domain index?

jhauser-us, 2025-05-02:
> Then how to find the mapping betwen child index and the real domain index?

As far as the AIA goes, this is entirely implementation-specific.

I believe most existing APLIC implementations have only a single M-level interrupt domain (the root domain) and a single child S-level interrupt domain (child index 0 only).  Unless you know you really need multiple child domains, there's no reason to build anything more complex.

(A future "Supervisor Domains" RISC-V extension is proposing to add some rules about an APLIC's supervisor-level interrupt domains when that extension is also implemented.)

Yasinhwang, 2025-05-06:
OK, I got it, thanks.

closed 2025-05-06 by Yasinhwang