riscv/riscv-aia #124: vu mode

shengqilong, 2025-04-22:
"In particular, when vsiselect has a reserved value, attempts to access sireg from a virtual machine (VS or VU-mode) should preferably raise an illegal instruction exception instead of a virtual instruction exception."

“A virtual instruction exception is raised for attempts from VS-mode or VU-mode to directly access vsireg, or attempts from VU-mode to access sireg.”

There seems to be a conflict between the above two paragraphs in spec

radimkrcmar, 2025-04-22:
The AIA spec does not define the general behavior of siselect and sireg anymore -- they got ratified as Sscsrind.

The manual states that sireg accesses from VU-mode must raise a virtual instruction exception:
```
A virtual instruction exception is raised for attempts from VS-mode or VU-mode to directly access
vsiselect or vsireg*, or attempts from VU-mode to access siselect or sireg*.

The behavior upon accessing vsireg* from M-mode or HS-mode, or accessing sireg* (really vsireg*)
from VS-mode, while vsiselect holds a value that is not implemented at HS level, is UNSPECIFIED.
```

> "In particular, when vsiselect has a reserved value, attempts to access sireg from a virtual machine (VS or VU-mode) should preferably raise an illegal instruction exception instead of a virtual instruction exception."

This sentence is a meaningless changelog, but I agree that it incorrectly summarizes the changes.

shengqilong, 2025-04-22:
Thank you. During implementation, if vsiselect is a reserved value, accessing sireg (which is actually vsireg) in VS mode triggers an illegal instruction. Is this acceptable? If not, are there any better alternatives or recommendations for handling such cases?

radimkrcmar, 2025-04-22:
It is acceptable and the manual even recommends it.

Just remember that hstateen0[60] still takes precedence and turns every VS sireg access into virtual instruction exception.

shengqilong, 2025-04-23:
If the Smaia/Ssaia extension is implemented, can the Smcsrind/Sscsrind extension be omitted? If so, would it be unnecessary to implement hstateen0[60] (CSRIND) ? Would the following exception then not need to be implemented?

```
If the CSRIND bit is one in mstateen0 but is zero in hstateen0, then all attempts from VS or VU-mode to access 
siselect or sireg raise a virtual instruction exception, not an illegal instruction exception, regardless of the value
of vsiselect or any other mstateen bits.
```

radimkrcmar, 2025-04-23:
The manual looks like Smaia/Ssaia depends on Smcsrind/Sscsrind.

Are you planning to implement AIA without IMSIC?

I agree the manual does not allow the simplest implementation in this case, but probably the only option now is to add new extension name for Ssaia without IMSIC.

shengqilong, 2025-04-23:
Given that both AIA and IMSIC must be implemented, is it mandatory to support the Smcsrind/Sscsrind extensions under these circumstances? However, the following explanations have left me very confused.

```
Smcsrind/Sscsrind is an ISA extension that extends the indirect CSR access mechanism originally defined as 
part of the [Smaia/Ssaia extensions](https://github.com/riscv/riscv-aia), in order to make it available for use 
by other extensions without creating an unnecessary dependence on Smaia/Ssaia.
```
Besides, the Smcsrind/Sscsrind extensions contain *ireg2~6. In the implementation, these registers won't exist.

radimkrcmar, 2025-04-23:
I see, I correct my previous statement, sorry.  Smaia/Ssaia doesn't depend on Smcsrind/Sscsrind -- the common subset is just defined twice, and must be consistent in both definitions.

* You don't have to implement *ireg2-6 with Smaia/Ssaia.
* You have not implemented Smcsrind/Sscsrind unless you have mireg2-6/sireg2-6.
* You still have to implement hstateen[60] for Ssaia if you also have implemented Sstateen.

shengqilong, 2025-04-24:
Thank you very much. I have another question: If only the Ssstateen extension is implemented (without implementing the Smstateen extension), then mstateen0 does not exist, and only sstateen0 and hstateen0 are present. In this case, when in VS mode, if hstateen0[60] = 0, will accessing sireg (which actually refers to vsireg) still trigger a virtual exception?

radimkrcmar, 2025-04-27:
Yes.

```
Extension Ssstateen is defined as the supervisor-level view of Smstateen. Therefore, the combination
of Ssaia and Ssstateen incorporates the bits defined above for hstateen0 but not those for mstateen0,
since machine-level CSRs are not visible to supervisor level.
```

closed 2025-05-01 by shengqilong