riscv/riscv-aia #125: What happens when M/HS modes access vsireg while vsiselect points to a reserved region?

shengqilong, 2025-04-27:
When vsiselect points to a reserved region (addresses in the ranges 0x00–0x2F, 0x40–0x6F, 0x100–0x1FF, or greater than 0x1FF), will accessing the vsireg register in M-mode/HS-mode trigger an exception, and if so, what type of exception will it be?

radimkrcmar, 2025-04-27:
UNSPECIFIED.  Most likely illegal instruction exception.  The information about implementation should be given to software.

```
The behavior upon accessing vsireg* from M-mode or HS-mode, or accessing sireg* (really vsireg*)
from VS-mode, while vsiselect holds a value that is not implemented at HS level, is UNSPECIFIED.

  (i)    It is recommended that implementations raise an illegal instruction exception for
  (i)    such accesses, to facilitate possible emulation (by M-mode) of these accesses.
```

jones-drew, 2025-04-28:
It'd be nice if Ssstrict of the rva23 profile spec covered this, but that's not exactly spelled out.

closed 2025-05-01 by shengqilong