riscv/riscv-aia #126: Some questions about indirect access

Baixyzz, 2025-04-28:
From privileged architecture,we can see:

> These conditions for an interrupt trap to occur must be evaluated in a bounded amount of time from when an interrupt becomes, or ceases to be, pending in mip, and must also be evaluated immediately following the execution of an xRET instruction or **an explicit write to a CSR on which these interrupt trap conditions expressly depend (including mip, mie, mstatus, and mideleg).**

When we write the interrupt-file registers through indirect access via *iselect CSR and its partner *ireg CSR, do we need to immediately re-evaluate whether the interrupt will cause a trap? 
Furthermore, can we assume that the next instruction after the indirect access can read the *ip CSR to get the result of writing the interrupt-file registers through indirect access?

Baixyzz, 2025-05-06:
Hi @jhauser-us ,sorry to bother you.About these questions,can you give us some advice?

jhauser-us, 2025-05-09:
The AIA document was recently clarified to say
> Changes to the state of an interrupt file are guaranteed to be reflected in the relevant interrupt-pending bit in CSR `mip` or `hgeip` eventually, but not necessarily immediately.

Because of this potential delay, the answer to both your questions is "no".

(The text you quoted from the base Privileged spec ought to be improved to be a little clearer about the cases where delays are explicitly allowed in the propagation of interrupt signals.)

Baixyzz, 2025-05-09:
Thanks for you reply.I will open an issue in the base Privileged spec to make it a little clearer.

closed 2025-05-09 by Baixyzz