riscv/riscv-aia #137: Which execption should be rasied when write vstopi in VS-mode?

Yasinhwang, 2025-08-11:
vstopi is a HRO CSR, read vstopi in VS-mode will raise illegal instruction excepiont, how about the write access,  if written in VS-mode ,whch exception should be raised? Illegal instruction excepiont or virtual instruction excepion?

yjloong, 2025-08-12:
> riscv-interrupts-20250312.pdf
> 2.4. Virtual instruction exceptions
> Following the default rules for the H extension, attempts from VS-mode to directly access a hypervisor or VS CSR other than vsireg, or from VU-mode to access any supervisor-level CSR (including  hypervisor and VS CSRs) other than sireg or vsireg, usually raise not an illegal instruction exception but instead a virtual instruction exception. 

> riscv-privileged.pdf
>  2.1. CSR Address Mapping Conventions
>  Instructions that access a non-existent CSR are reserved. Attempts to access a CSR without appropriate privilege level raise illegal-instruction exceptions or, as described in Section 19.6.1, virtual-instruction exceptions.  Attempts to write a read-only register raise illegal-instruction exceptions. 

I think the access to VS CSRs is privilege question first.
So read/write vstopi in VS-mode should raise virtual instruction exception for HS-mode.

Just for reference.

Yasinhwang, 2025-08-14:
AIA spec chapter 2.4 says:

[image]

For hypervisor CSR other than vsireg, access from VS-mode usually raise virtual instruction execption,  but the vstopi is a HRO CSR, vstopi is not the usual case, rule rely on RISC-V Privileged spec:

[image]
vstopi is  simillar with the CSR hgeip, both the two CSRs are HRO, they should first be  HS-qualified, then they can be reported with a virtual instruction execption. Wirte HRO  is not HS-qualified, so I think write vstopi should raise illegal-instruction, but I need a confirmation.